CCNYRoboticsLab/scan_tools #19: Make laser_scan_matcher work with scans of more/less than 180 degrees

corot, 2015-03-03:
As explained in [this ROS answer](http://answers.ros.org/question/203929/does-laser_scan_matcher-work-with-scans-of-moreless-than-180-degrees/), it seems to me that laser_scan_matcher doesn't work with scans of more/less than 180 degrees. Would be a great improvement to allow this, for example for TurtleBots that use a "laser scan" of only 55 deg. faked from kinect depth images.

awesomebytes, 2015-03-15:
I've been using a fake laser from a depthimage_to_laserscan source from a structure.io and it kind of works for a while... until the laser data seems to not be enough. I mean, there is really just a few laser points to match, I think it makes sense. (57º FOV...)
This is my launch file:
https://gist.github.com/awesomebytes/73e2d5e3e7bcb0457d24

It includes the use of imu and vel (this last one made my tests work way better surprisingly).

corot, 2015-04-24:
Hi, I have retried the scan matcher with your parameters and performance is notably better, regardless the fov of the laser (my bad... i blamed the wrong guy). Thanks!!!

closed 2015-04-24 by corot