CCNYRoboticsLab/scan_tools #23: plan for releasing deb package

k-okada, 2015-07-08:
Hi,

I would these packages are super useful, do you have any plan to release these package as deb files? so we can use them just install via apt-get

130s, 2015-07-08:
+1 for Indigio where it builds w/o any issues for me.

130s, 2015-07-14:
@idryanov @miguelsdc if you all are busy I can help to make a new release into newer ROS distros (maybe Indigo, Jade. Hydro is end-of-life).

slivingston, 2015-09-23:
I also can help. Has anyone had contact with the repo owners?

ubuntuslave, 2015-09-24:
Try contacting the author, ivan.dryanovski@gmail.com

Good luck!

130s, 2015-09-24:
According to [MaintenanceGuide](http://wiki.ros.org/MaintenanceGuide#Claiming_Maintainership) at ros.org, contacting via issue should be enough (I'm afraid we should better refrain from citing someone's personal email publicly..). Next we may ask for help at ros-release.

ubuntuslave, 2015-09-25:
Oops. Sorry, I thought his email was publicly available on the repos (for instance in CCNY RGB-D tools) any way.

Carlos

130s, 2015-10-05:
[Asked a guidance](http://lists.ros.org/pipermail/ros-release/2015-October/004695.html) at ros-release on what to do to move forward.

ubuntuslave, 2015-10-05:
I'm willing to help, but I have no more Linux computers with the latest Ubuntu nor ROS on them. I really stopped using ROS a couple of years ago because I was tired of dual booting or using virtual machines. 

@130s Let's talk privately to set up the distro and get it working with the latest ROS versions and as debian package. I think @idryanov is busy now working with the Tango project at Google, so please forgive us for taking too long to respond. Contact me a ubuntuslave AT Ge MALE . Kom

130s, 2015-10-13:
@ubuntuslave Progress made by @wjwwood at [ML](http://lists.ros.org/pipermail/ros-release/2015-October/004698.html) and seems like you have a write access. Could you move forward make a new release?

ubuntuslave, 2015-10-13:
I added @130s and @wjwwood as collaborators to the scan_tools repo. Let me know how else I can help.

wjwwood, 2015-10-13:
I don't actually have time to administrate the repository, but I'm glad to help on specific tasks when asked to.

ubuntuslave, 2015-10-13:
Can @130s help us by making "a new release into newer ROS distros "?

130s, 2015-10-14:
@ubuntuslave Sure. I'm waiting for https://github.com/ccny-ros-pkg/scan_tools/issues/26 to move forward.

130s, 2015-10-14:
Done for Indigo https://github.com/ros/rosdistro/pull/9652

closed 2015-10-14 by 130s

k-okada, 2015-10-15:
Thanks you!!! :smile: