CCNYRoboticsLab/scan_tools #26: [Suggestion] indigo branch

130s, 2015-10-14:
Given that this package depends on some libraries that their release conditions tend to vary per Ubuntu and ROS distro (e.g. `pcl`), I suggest setting a default branch that couples with ROS distros.
- For now, ROS Indigo should be good
- Since there already branches exist for older distros (fuerte, hydro), naming it as `indigo` 
 - We can release from `indigo` branch into both Indigo and Jade until any issue comes up

@ubuntuslave If this sounds fine can you create a new branch and set it as a repository default?

ubuntuslave, 2015-10-14:
Why not just leave the `master` as the default and rebase it with the `indigo` branch?

ubuntuslave, 2015-10-14:
@130s I branched a new `indigo` branch from `master`.

130s, 2015-10-14:
> Why not just leave the master as the default and rebase it with the indigo branch?

So that handling an issue like https://github.com/ccny-ros-pkg/scan_tools/pull/24 becomes easier; For example, if we merge https://github.com/ccny-ros-pkg/scan_tools/pull/24 into `master` (assuming my comment https://github.com/ccny-ros-pkg/scan_tools/pull/24#issuecomment-148068451 was right that the PR #24 works only for post-Hydro), people who still want to use Hydro can't build from master out of the box, but they still can using `hydro` branch that already exists.

(Note that separating distro(s)-specific branches may require backporting commits that work in earlier distros, if we like to maintain the same feature over branches. This may depend on requests from users IMO. I know at some repositories maintainers make great effort to make `master` compatible to various distros, but I must say it seems it gets costly in general)

Hope my explanation is clear.

130s, 2015-10-14:
And for default branch, I also insist on making the latest as the default branch. That way we can reduce the effort of maintaining branches (same is done at many ROS repositories).

ubuntuslave, 2015-10-15:
@130s Ok. I agree. Forgive my ignorance: I have been out of the ROS loop for a couple of years so I'm trying to get back to the game. I made the `indigo` branch the default branch of this repo.

closed 2015-10-15 by ubuntuslave

130s, 2015-10-15:
No need to be sorry at all! These things are not documented anywhere, only repository stalkers know :palm_tree: