CCNYRoboticsLab/scan_tools #30: ROS Buildfarm build failure "fatal: unable to access 'https://github.com/AndreaCensi/csm.git/': Problem with the SSL CA cert"

130s, 2015-10-19:
Reported at https://github.com/ccny-ros-pkg/scan_tools/pull/29#issuecomment-149214344.

Jenkins build fails http://jenkins.ros.org/job/ros-indigo-laser-scan-matcher_binarydeb_trusty_amd64/5/console when it tries to download `csm` from other github repo.

```
make -f CMakeFiles/Makefile2 all
make[3]: Entering directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
make -f CMakeFiles/csm_src.dir/build.make CMakeFiles/csm_src.dir/depend
make[4]: Entering directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
cd /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450 /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450 /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/CMakeFiles/csm_src.dir/DependInfo.cmake --color=
Scanning dependencies of target csm_src
make[4]: Leaving directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
make -f CMakeFiles/csm_src.dir/build.make CMakeFiles/csm_src.dir/build
make[4]: Entering directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
/usr/bin/cmake -E cmake_progress_report /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/CMakeFiles 3
[  9%] Creating directories for 'csm_src'
/usr/bin/cmake -E make_directory /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/src
/usr/bin/cmake -E make_directory /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/src
/usr/bin/cmake -E make_directory /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm
/usr/bin/cmake -E make_directory /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/tmp
/usr/bin/cmake -E make_directory /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/stamp
/usr/bin/cmake -E make_directory /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/src
/usr/bin/cmake -E touch /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/stamp/csm_src-mkdir
/usr/bin/cmake -E cmake_progress_report /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/CMakeFiles 4
[ 18%] Performing download step (git clone) for 'csm_src'
cd /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm && /usr/bin/cmake -P /tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu/csm/tmp/csm_src-gitclone.cmake
Cloning into 'src'...
fatal: unable to access 'https://github.com/AndreaCensi/csm.git/': Problem with the SSL CA cert (path? access rights?)
Cloning into 'src'...
fatal: unable to access 'https://github.com/AndreaCensi/csm.git/': Problem with the SSL CA cert (path? access rights?)
Cloning into 'src'...
fatal: unable to access 'https://github.com/AndreaCensi/csm.git/': Problem with the SSL CA cert (path? access rights?)
-- Had to git clone more than once:
          3 times.
CMake Error at tmp/csm_src-gitclone.cmake:40 (message):
  Failed to clone repository: 'https://github.com/AndreaCensi/csm.git'

make[4]: *** [csm/stamp/csm_src-download] Error 1
make[4]: Leaving directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
make[3]: *** [CMakeFiles/csm_src.dir/all] Error 2
make[3]: Leaving directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450/obj-x86_64-linux-gnu'
dh_auto_build: make -j1 returned exit code 2
make[1]: *** [override_dh_auto_build] Error 2
make[1]: Leaving directory `/tmp/buildd/ros-indigo-laser-scan-matcher-0.2.1-0trusty-20151019-0450'
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
E: Failed autobuilding of package
W: no hooks of type C found -- ignoring
I: unmounting /var/cache/pbuilder/ccache filesystem
I: unmounting dev/pts filesystem
I: unmounting proc filesystem
I: cleaning the build env 
I: removing directory /var/cache/pbuilder/build//21067 and its subdirectories
Process leaked file descriptors. See http://wiki.jenkins-ci.org/display/JENKINS/Spawning+processes+from+build for more information
Build step 'Execute shell' marked build as failure
Build step 'Groovy Postbuild' marked build as failure
Build step 'Groovy Postbuild' marked build as failure
Description set: 
Sending e-mails to: ros-buildfarm-release@googlegroups.com ccnyroboticslab@gmail.com cjaramillo@gc.cuny.edu
Finished: FAILURE
```

130s, 2015-10-19:
I was already kind of afraid of, but ROS facility might not be permissible of the way currently `laser_scan_matcher` builds CSM (i.e. downloading source code of CSM during build process). So I will move forward to release `CSM` as a ROS Third Party Package (see [wiki](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty#preview)).

We have 2 choices at least:
- I've opened a PR against the upstream: CSM. https://github.com/AndreaCensi/csm/pull/10 If this gets accepted, I'll make a release of CSM as a ROS pkg and then add necessary change to laser_scan_matcher, and make a new release.
- It's possible the maintainer of CSM is busy and won't merge the PR above soon, but we have to fix the buildfarm issue soon (since this will block new ROS public sync). In that case I'll take a 2nd option mentioned in ROS wiki, "[Modifying_the_Release_Repository](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty#Modifying_the_Release_Repository)". This is going to be an interim solution until the PR against CSM above will be merged.

AndreaCensi, 2015-10-19:
The PR was accepted quite rapidly this time :-)

130s, 2015-10-19:
Thanks to @AndreaCensi's prompt response, I'm making a progress. Now CSM release request to ROS Indigo is in flight https://github.com/ros/rosdistro/pull/9677.

130s, 2015-10-21:
Ok, [CSM deb is already built and available](http://www.ros.org/debbuild/indigo.html?q=csm) on ROS [Shadow repo](http://wiki.ros.org/ShadowRepository), but I forgot to add build and install rules (https://github.com/AndreaCensi/csm/pull/11) so what its deb does currently is only to install package.xml (silly me). Unfortunately I won't be able to work on this until this weekend at earliest. So if someone can help adding change to CSM (`csm_eigen` branch) I appreciate it very much.

130s, 2015-10-27:
I'm hoping to take some time this weekend (yes I'm saying it again) to work on this.

Meanwhile @miguelsdc you seem to have [catkinized a portion where csm gets downloaded](https://github.com/ccny-ros-pkg/scan_tools/blob/indigo/laser_scan_matcher/csm/GetCSM.cmake). Would you be willing to help writing [CMakeLists.txt on csm_eigen branch in csm](https://github.com/AndreaCensi/csm/blob/csm_eigen/CMakeLists.txt)?

k-okada, 2015-11-02:
> but ROS facility might not be permissible of the way currently laser_scan_matcher builds CSM (i.e. downloading source code of CSM during build process)

it is not a good, but better than leave broken package, so if I ware maintainer hee, I would choose adding depends to wget or ca_certificates, but menteiners here already choose right but difficult path; here is the solution https://github.com/ccny-ros-pkg/scan_tools/pull/31

csm deb is also not working right now -> https://github.com/tork-a/csm-release/issues/1

130s, 2015-11-11:
Finally release request for a new version is sent https://github.com/ros/rosdistro/pull/9806. 
Closing this ticket hoping the new release will fix this.

closed 2015-11-11 by 130s