CCNYRoboticsLab/scan_tools #39: Cannot build newer versions that don't compile CSM

tlalexander, 2015-12-17:
Some time around commit d214d705525717e9dac93b2a1a0ff4e1e35c9e46, the function to automatically compile CSM was removed.

Both myself and two other coworkers have tried to compile CSM and link it to this library with no success. I'm not sure what I'm doing wrong, but it was a big pain to try. Ultimately, I switched to the above commit, where CSM is automatically downloaded and compiled, and everything magically worked.

Given that this library depends on CSM, I wonder if it would be worthwhile to put that functionality back, maybe with a command line switch? Alternatively, some suggestions on a known method for compiling and linking with CSM in the documentation would be helpful.

130s, 2015-12-17:
Sorry for the inconvenience. You can now install `laser_scan_matcher` via `apt-get`, by `sudo apt-get install ros-indigo-scan-tools` (for Indigo).

If you have specific reason to build from source, `git clone https://github.com/AndreaCensi/csm.git` into your catkin workspace and build together (I'll have to document this).

Please reopen if the above is not sufficient.

closed 2015-12-17 by 130s

130s, 2015-12-18:
> (I'll have to document this).

I've made some update. https://github.com/ccny-ros-pkg/scan_tools/pull/40 Hope this helps.

tlalexander, 2015-12-18:
Thanks for the quick response. I'm using ROS Jade, which does work with the version of scan_tools I mention above when build from source. There are no binaries for Jade in the repo.

The new instructions do not work with Jade. I when I try to run the wstool line, I get the following error:
> ERROR in config: /home/taylor/catkin_test/src has no workspace configuration file '.rosinstall'

If I clone the repo directly into the src directory of my catkin workspace, run the rosdep line (with jade instead of indigo) and then run catkin_make, I get an error that it couldn't find csm. So it seems csm is not automatically compiled in this configuration.

I am on Ubuntu 15.04.

130s, 2015-12-18:
@tlalexander Sorry, https://github.com/ccny-ros-pkg/scan_tools/pull/41 is needed too.

> There are no binaries for Jade in the repo.

Their binary should be available. `apt-get install ros-jade-scan-tools`.

tlalexander, 2015-12-18:
Thanks, compiling from source seems to work now. However, ros-jade-scan-tools is not found in my repo. All the other ros-jade stuff is there.

130s, 2015-12-18:
> However, ros-jade-scan-tools is not found in my repo. All the other ros-jade stuff is there.

Oh, now I see what's going on. I'm working to release the missing packages into Jade.