CCNYRoboticsLab/scan_tools #42: Release into Jade

130s, 2015-12-18:
Pointed out https://github.com/ccny-ros-pkg/scan_tools/issues/39#issuecomment-165867753.

https://github.com/ros/rosdistro/pull/9733#issuecomment-152684150 removed `scan_tools` meta package and `laser_scan_matcher` from Jade. They should be now good to be re-released (they already are re-released into Indigo).

130s, 2015-12-18:
CSM is in the air https://github.com/ros/rosdistro/pull/10092

130s, 2015-12-22:
Done https://github.com/ros/rosdistro/pull/10120

closed 2015-12-22 by 130s

130s, 2015-12-24:
@tlalexander DEB of `scan_tools` for Jade is now available in so-called [Shadow Repository](http://wiki.ros.org/ShadowRepository). Nevermind if you have no idea what it is; it'll still become available soon when "public repository sync" happens (that's when the ROS manager announces new packages ([e.g. this](http://lists.ros.org/pipermail/ros-users/2015-November/069765.html)).

tlalexander, 2016-01-05:
Thanks! I see that it is still not in my repo, but I will wait a few more weeks and see if it gets sync'd.