CCNYRoboticsLab/scan_tools #47: package 'openni-dev' not found

libing64, 2016-04-12:
In fact, I have installed libopenni-dev and libopenni2-dev, but it always said "package openni-dev not found".

my ros version: indigo

130s, 2016-04-12:
Please provide reproducible scenario, and the full length output. I suspect otherwise very few people can try tracking down the issue you're seeing.

libing64, 2016-04-12:
@130s 

-- +++ processing catkin metapackage: 'scan_tools'
-- ==> add_subdirectory(scan_tools)
-- +++ processing catkin package: 'laser_scan_sparsifier'
-- ==> add_subdirectory(laser_scan_sparsifier)
-- +++ processing catkin package: 'laser_scan_splitter'
-- ==> add_subdirectory(laser_scan_splitter)
-- +++ processing catkin package: 'ncd_parser'
-- ==> add_subdirectory(ncd_parser)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'laser_ortho_projector'
-- ==> add_subdirectory(laser_ortho_projector)
-- Using these message generators: gencpp;genlisp;genpy
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- +++ processing catkin package: 'laser_scan_matcher'
-- ==> add_subdirectory(laser_scan_matcher)
-- Using these message generators: gencpp;genlisp;genpy
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'csm'
--   package 'csm' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337 (_pkg_check_modules_internal)
  laser_scan_matcher/CMakeLists.txt:21 (pkg_check_modules)

-- +++ processing catkin package: 'polar_scan_matcher'
-- ==> add_subdirectory(polar_scan_matcher)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'scan_to_cloud_converter'
-- ==> add_subdirectory(scan_to_cloud_converter)
-- Using these message generators: gencpp;genlisp;genpy
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Configuring incomplete, errors occurred!
See also "/home/libing/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/libing/catkin_ws/build/CMakeFiles/CMakeError.log".

libing64, 2016-04-12:
libing@libing0:~/catkin_ws/scan_tools$ dpkg -l | grep openni

ii  libopenni-dev                                         1.5.4.0-7                                           amd64        headers for OpenNI 'Natural Interaction' frameworks
ii  libopenni-sensor-pointclouds0                         5.1.0.41.1-1                                        amd64        Microsoft Kinect sensor modules for the OpenNI framework
ii  libopenni0                                            1.5.4.0-7                                           amd64        framework for sensor-based 'Natural Interaction'
ii  libopenni2-0                                          2.2.0.33+dfsg-1~ubuntu14.04.1                       amd64        framework for sensor-based 'Natural Interaction'
ii  libopenni2-dev                                        2.2.0.33+dfsg-1~ubuntu14.04.1                       amd64        headers for OpenNI 'Natural Interaction' frameworks
ii  openni-utils                                          1.5.4.0-7                                           amd64        debug and test utilities OpenNI framework
ii  ros-indigo-freenect-launch                            0.4.1-0trusty-20160201-071140-0800                  amd64        Launch files for freenect_camera to produce rectified, registered or disparity images. Also produce point clouds and registered point clouds. Based on the openni_launch package.
ii  ros-indigo-openni2-camera                             0.2.5-0trusty-20151111-0829-+0000                   amd64        Drivers for the Asus Xtion and Primesense Devices. For using a kinect with ROS, try the freenect stack
ii  ros-indigo-openni2-launch                             0.2.2-0trusty-20160201-071304-0800                  amd64        Launch files to start the openni2_camera drivers using rgbd_launch.

libing64, 2016-04-12:
If the laser_scan_matcher is added to the black list, then the scan_tools can be compiled successfully.
-DCATKIN_BLACKLIST_PACKAGES="laser_scan_matcher"

130s, 2016-04-12:
> Please provide reproducible scenario, 

By this I meant the commands you used. 

Anyways, without fully knowing your conditions and thus not being able to replicate the issue, what if you run:

```
rosdep install --from-paths src --ignore-src --rosdistro indigo -y
```

and then build again?

(I don't know if `~/catkin_ws/scan_tools` works. I thought Catkin look for source only in `src` folder.)

BTW, you know binary for `scan_tools` is available right? `sudo apt-get install ros-indigo-scan-tools`.

libing64, 2016-04-12:
@130s Thanks for your kind reply. 

==
cd ~/catkin_ws
git clone xxx
rosdep install --from-paths scan_tools  --ignore-src --rosdistro indigo -y
catkin_make --source scan_tools
==
I know ros-indigo-scan-tools exits, but I need to modify the code so I should compile from source.

130s, 2016-04-13:
>  git clone xxx

What URL did you use? Please share *exactly* what you do as much as possible.

And assuming the URL in, I just did the following that builds fine.
```
cd ~/catkin_ws/
rm -fr build/ devel/
git clone https://github.com/ccny-ros-pkg/scan_tools
rosdep install --from-paths scan_tools --ignore-src --rosdistro indigo -y
catkin_make --source scan_tools
```

So again, we need the list of full, exact commands you use.

libing64, 2016-04-14:
@130s 
"git clone xxx" means
git clone https://github.com/ccny-ros-pkg/scan_tools

130s, 2016-04-14:
I see. 
As in https://github.com/ccny-ros-pkg/scan_tools/issues/47#issuecomment-209558058, I'm not able to reproduce your issue. And I now suspect your issue may or may not be scan_tools specific. I suggest you asking on http://answers.ros.org/ and paste the link to your question here so that we can join.

closed 2016-04-14 by 130s

libing64, 2016-05-23:
After installing ros-indigo-csm, the problem is solved.
`sudo apt-get install ros-indigo-csm`

130s, 2016-05-31:
> After installing ros-indigo-csm, the problem is solved.
> `sudo apt-get install ros-indigo-csm`

That must have been taken care of if you ran `rosdep` command I mentioned earlier.

Maybe sharing your output of ` rosdep install --from-paths scan_tools --ignore-src --rosdistro indigo -y` without `ros-indigo-csm` installed helps (if you still like to track down the issue).