CCNYRoboticsLab/scan_tools #48: scan matcher flag to prevent pose/tf publishing on pose jumps?

corot, 2016-04-19:
I'm using laser scan matcher combined with odometry using [robot_localization](https://github.com/cra-ros-pkg/robot_localization). In general works ok, but sometimes it does pose jumps without showing any error. This is different from issue #20: I don't experience that errors anymore. But jumps are by definition erroneous, as visual odometry must be continuous. So I propose to add a couple of parameters, for example min_jump_dist and min_jump_angle and ignore any pose update bigger than those values. Or alternatively, set a very big covariance proportional to the jump.

They can default to 0, meaning disable the feature.

What do you think? Make sense the idea?

1plus-1, 2020-02-05:
Lately I meet the similar issue. And I found the solution from #58. EwingKang had a fix in this fork: https://github.com/EwingKang/scan_tools. It fixes my issue.

130s, 2022-07-14:
@malban Do you think your https://github.com/CCNYRoboticsLab/scan_tools/pull/79 addresses this as well?

malban, 2022-07-15:
Not sure.  #79 fixes one major source of jumps in the current code, caused by incorrect predictions, which could have been the root cause of what this issue was addressing.  But jumps might also be caused by other things related to the sensor data and actual scan match failure.

Hard to know without more data or feedback from the reporter.