CCNYRoboticsLab/scan_tools #49: laser_scan_matcher license

Minipada, 2016-11-29:
Hello,

I was checking the license and there are 2 listed in the [package.xml](https://github.com/ccny-ros-pkg/scan_tools/blob/indigo/laser_scan_matcher/package.xml#L18). Which one is it?

Thanks!

130s, 2016-11-30:
Good question, I don't know the answer to that, and quick research didn't return specific answers either ([this page](https://fedoraproject.org/wiki/Packaging:LicensingGuidelines?rd=Packaging/LicensingGuidelines#Multiple_Licensing_Scenarios) was helpful but I'm still not clear what to do). If you could figure that out and let us know, that'll be appreciated.

A little background of the situation:
- `scan_tools` ROS package itself was written under `BSD`.
- [csm](http://wiki.ros.org/csm), which a part of scan_tools depends on, is `LGPL`.

130s, 2023-02-19:
Dup of https://github.com/CCNYRoboticsLab/scan_tools/issues/76

closed 2023-02-19 by 130s