CCNYRoboticsLab/scan_tools #50: Laser Ortho Projector

raynerthestoner, 2017-02-13:
Hi,

I realize the code requires the use of either an `imu` or a `geometry_msgs/pose`. I do not have an imu. May I know how to get the pose of my 2D RPLidar A2 sensor? The `pose` and `use_pose` is not documented anywhere in the package summary, so I have no idea how to make this node work properly as it is not providing the `base_ortho` tf transformation.

Thanks!

ubuntuslave, 2017-02-24:
you may just publish a [static tf](http://wiki.ros.org/tf#static_transform_publisher) for the `base_ortho` and set it to identity or to an estimated transform of your `laser_base_frame` or whatever it was called (I already forgot exactly). If you understand the transform links, then fixing the code to make it work without an IMU is possible although not intended for mapping nor height estimation because the relative pose of the laser scanner would be unknown.

raynerthestoner, 2017-02-27:
I am planning for my robot move on level ground but with a sensor tilted at a fixed pitch angle. If I have static transforms of `world to base link`  , `world->base_ortho` ,  `base_ortho->laser` ,  it should work right? Currently what I see on RVIZ is that the pointcloud2 is not accurately projected vertically downwards as shown in this [picture](https://www.dropbox.com/s/f477j0jghz6hb31/asas.png?dl=0). Is there a reason why it is like this?