CCNYRoboticsLab/scan_tools #51: Release for kinetic

stwirth, 2017-03-10:
Hi, could we get a release of scan_tools for ROS kinetic please?
csm has been released for kinetic recently https://github.com/ros/rosdistro/pull/14128.

Thanks!

130s, 2017-03-10:
`csm` for Kinetic has been failing to build on ROS buildfarm (e.g. [this](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__csm__ubuntu_xenial_amd64__binary/3/console)) so that should be fixed first. 
I won't be able to work on that for a few weeks at least. If someone can give it a shot that'll be great.

stwirth, 2017-03-20:
latest build for csm succeeded: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__csm__ubuntu_xenial_amd64__binary/6/

130s, 2017-03-22:
https://github.com/ros/rosdistro/pull/14260 Done

closed 2017-03-22 by 130s