CCNYRoboticsLab/scan_tools #54: Laser Scan Matcher Compatibility

HUDiwan, 2017-05-01:
Hello, I am trying to use the laser scan matcher in Kinetic - Ubuntu 16.04. However, the laser scan matcher node is not running. I believe it is compatibility issue, can you please verify?
Thanks.

EwingKang, 2018-03-04:
@HUDiwan  I can run the node on 16.04+Kinetic without problem( well, except the odometry bug that is)
Have you install CSM?

trixr4kdz, 2018-04-05:
Hi @EwingKang, I'm also trying to do the same thing, but it seems like I can't get the `odom->base_link` tf. I'm using `tf2_ros` to publish the `base_link->laser_frame` and on my launch file for laser_scan_matcher_node, I have `fixed_frame = odom` and `base_frame = base_link` and I've remapped the `scan` topic to the correct topic. How did you manage to get yours to work?