CCNYRoboticsLab/scan_tools #59: laser_scan_matcher subscribing to wrong message type: nav_msgs/Odometry/

GarvSharma, 2018-03-18:
Hello,
I am trying to use laser scan matcher for scan-based localization, on **ROS Indigo**. 
However, when laser_scan_matcher starts, I get an error;
**[ERROR] [1521390276.090566633]: Client [/laser_scan_matcher_node] wants topic /map to have datatype/md5sum [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7], but our version has [nav_msgs/OccupancyGrid/3381f2d731d4076ec5c71b0759edbe4e]. Dropping connection.**

It seems that laser_scan_matcher is trying to subscribe to /map topic but with wrong message type: **nav_msgs/Odometry**, instead of what the system is publishing: **nav_msgs/OccupancyGrid** which actually contains map data. These are two entirely different definitions.

How do I correct this? Due to this, I have no map < odom transform in my tf tree and I get the error: 
[ WARN] [1521390281.116509155]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.101214 timeout was 0.1.

Thanks

GarvSharma, 2018-03-18:
So, I changed:
`if (!nh_private_.getParam ("use_odom", use_odom_))
    use_odom_ = true;`
to:
`if (!nh_private_.getParam ("use_odom", use_odom_))
    use_odom_ = false;`

I do not get the datatype/md5sum error, but the:
`[ WARN] [1521390281.116509155]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.101214 timeout was 0.1.`
is still there and no map < odom transform in tf view_frames

rivkaer, 2018-11-23:
Have you solved this problem?