CCNYRoboticsLab/scan_tools #60: Error in scan matching

af-silva, 2018-04-19:
Hi, 

I have a robot application and I was testing this package for Laser Odometry and stumbled with this problem error running the node, I have briefly looked into the code but I was not able to understand the cause of the error.

`:err: Valid: 27/1083 invalid: 1056.
[ WARN] [1524149236.189355990, 17446.189000000]: Error in scan matching`

So, my software configuration is as follows: I have a node publishing the laser scanner of to other lasers to a frame_id (the laser frame) with a transform (tf) to the **base_link**, and my base frame is the **odom** (instead of the world). I have also the **map** (that is used and published by the AMCL Package).

TFs: 
map -> odom
odom -> base_link
base_link -> laser

The laser scanner is being published in the **scan** topic.

Anyone have an idea why this happens, I would appreciate any help regarding this, 
In the mean time I will continue searching for possible solution. 

Thanks in advance,

ClaraGomez, 2019-09-09:
Hi, I am having the same problem. Did you find a solution? Thanks!

af-silva, 2019-09-16:
Hi @ClaraGomez, sorry for the late reply... 
This was sometime ago and I didn't pursued a solutions for this, I couldn't understand the cause, so I switch to another solution, try [this one](http://wiki.ros.org/rf2o_laser_odometry).

Sorry for not being much of a help ;)