CCNYRoboticsLab/scan_tools #63: Can't build laser_scan_matcher for melodic

msadowski, 2019-03-18:
Hi! I've been trying to build the package from source for melodic but can't seem to get it right.
Running `catkin build laser_scan_matcher` I see that csm builds OK (there are some warnings but it goes through) but I get an error from the laser_scan_matcher:

```
Starting  >>> laser_scan_matcher                  
_______________________________________________________________________________________________________________
Errors     << laser_scan_matcher:cmake /home/mat/.../ros_ws/rr_dev/logs/laser_scan_matcher/build.cmake.000.log
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  CMakeLists.txt:21 (pkg_check_modules)
```

The CmakeLists.txt the code is pointing to is: `pkg_check_modules(csm REQUIRED csm)`

Running `rospack find csm` returns the correct path.

Is there anything obvious I'm missing?

FranekStark, 2019-04-24:
Hey, i had the same problem and this is my solution:

1. clone the csm-Repo in your _catkin_ws_: `git clone https://github.com/AndreaCensi/csm`
2. move in to the repo `cd csm`
3. move back `cd ..`
4. solve dependencies: `rosdep install --from-paths src --ignore-src -r -y`
5. move into csm `cd csm`
6. Use the install instructions from their [manual](file:///C:/Users/Franek/Downloads/csm_manual.pdf):
```
cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr/local .
make
sudo make install
```
7. Move back `cd ..`
8. Install `catkin_make`

I hope, that will solve your Problem!

Lyauk, 2019-07-31:
Hey, try my metod
1  cd ~/catkin_ws/src
2  git clone https://github.com/AndreaCensi/csm
3  cd ..
4 catkin_make_isolated
5 cd ~/catkin_ws/src
6 git clone https://github.com/ccny-ros-pkg/scan_tools.git
7 cd ..
8 catkin_make_isolated
9 source devel/setup.bash

rfzeg, 2020-09-11:
(ROS Melodic)
I am able to clone and build csm, but when trying to build scan_tools I get this error message:

-- Checking for module 'csm'
--   No package 'csm' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  CMakeLists.txt:21 (pkg_check_modules)

-- Configuring incomplete, errors occurred!

<== Failed to process package 'laser_scan_matcher': 

I really appreciate your help.

jremy97, 2021-02-23:
(ROS Melodic)

@Lyauk I used your method to build the package and it succeeded, but when I tried to use it in a test launch file, I get the following error:

ERROR: cannot launch node of type [laser_scan_matcher/laser_scan_matcher_node]: Cannot locate node of type [laser_scan_matcher_node] in package [laser_scan_matcher]. Make sure file exists in package path and permission is set to executable (chmod +x)

 I am trying to use the laser scan matching package to localize a mmWave Sensor that outputs PointCloud2 messages

Any advice is greatly appreciated. Thank you!

AlperenKosem, 2021-03-07:
@jremy97 i had same issue and i solved like below.

[https://github.com/ccny-ros-pkg/scan_tools/issues/63#issuecomment-486194571](url)  
i followed these instructions. You should careful location of csm file it should be in your workspace not in src directory. 
After that, 

1 -  cd ~/catkin_ws/src
2 -  git clone https://github.com/ccny-ros-pkg/scan_tools.git
3 - catkin_make
4 - source devel/setup.bash

130s, 2023-02-19:
https://github.com/CCNYRoboticsLab/scan_tools/issues/63#issuecomment-516967220 seems to be a workaround many people found useful.

Closing.

On a related note, unless you need to build the packages by yourself (e.g. you need to modify the code of `csm` or `scan_matcher`, or you're on non Debian OS etc.), you should use pre-built packages that are available from `apt` package manager.

closed 2023-02-19 by 130s