CCNYRoboticsLab/scan_tools #64: Using laser_scan_matcher to find transforms between 2 robots with lidar.

Xvdgeest, 2019-11-29:
Hi, 

I was wondering if I could use the CSM to estimate the transform between two robots, both equipped with Lidar, scanning their surroundings (and each other). The starting positions will be unknown.

I guess I would have to adjust the laser_scan_matcher_node to accept two /scan topics, use the CSM to match both scans and get a transform. 

I'd like hear your opinion on this.

Cijoxx, 2019-11-30:
Hi,

Sorry for that I cannot help you ,i haven't used two robots in a system。But i have an immaturity idea for your question.First,you can drive two robot to get maps of the same surrounding(the larger of two maps).Then, let robots connect to the same master,set the tf, share their locations,complete the transform.

Xvdgeest, 2019-12-03:
Hi,

this is also an option I am investigating, using the multirobot map merge package to get reliable transforms. 
https://github.com/hrnr/m-explore/pull/30
However, since I don't need to build a map, I wanted to try with laser_scan_matcher as well.