CCNYRoboticsLab/scan_tools #67: Release for Melodic and Noetic

simonbogh, 2020-10-15:
Hi,

Are there any plans to update and release the package so it becomes available for Melodic and Noetic?

stwirth, 2021-02-15:
ping @130s

130s, 2021-02-15:
Ok...I'm no longer a user but have just started updating CI for M and N. https://github.com/ccny-ros-pkg/scan_tools/pull/68

130s, 2021-02-15:
- M: https://github.com/ros/rosdistro/pull/28331
- N: https://github.com/ros/rosdistro/pull/28332

Release will be completed when:
- Done for those who use from source, or on non-Ubuntu platform.
- For Ubuntu users who wish to use binary installer, once they are merged AND OSRF announces their apt repo is in sync.

Either way, I'm closing this as we're very close. Please feel free to leave comments for questions.

closed 2021-02-15 by 130s

stwirth, 2021-02-16:
Thank you @130s!