CCNYRoboticsLab/scan_tools #69: [Maintenance]: Maintainer in need

130s, 2021-02-15:
# Problem
I joined the maintenance of `scan_tools` repo when I was an active user and the maintainers back then seemed to be busy. Unfortunately I no longer am the user. AFAICT, other then-maintainers have not been active (checking in @ubuntuslave). So I'd call this repo is nearly unmaintained (I just made first releases into newer ROS distro https://github.com/ccny-ros-pkg/scan_tools/issues/67#issuecomment-779430590, where I claim as unmaintained).

# Discussion
Unless the current maintainers can be back active, I'd suggest calling for maintainership.

If no maintainer shows up, one thing to continue making the software alive is to register ["Orphaned packages"](http://wiki.ros.org/OrphanedPackage). With that, at least someone will continue making releases (review PRs, bugfix are not guaranteed IIUC).

130s, 2021-04-30:
Checking in again. If I hear none then I'll start "Orphaned package" process.

Care to chime in @ubuntuslave  or any others who've contributed to this repo?

sisaha9, 2021-07-05:
Maybe check over in ROS Discourse?

cst0, 2021-07-19:
Hey @130s -- I've started to rely on this package on a handful of projects here in the lab, am pretty comfortable with ROS, and have a small amount of experience pushing stuff to the ROS repositories. If you're still looking for a maintainer, I'm willing to volunteer.

solutiongeek2022, 2022-07-12:
The debugging of [laser beam welder](https://www.laserchina.com/welder/) makes sense.

130s, 2022-07-12:
@cst0 Sorry for the late response. Are you still interested in joining the maintainership?

@solutiongeek2022 I'm not sure how your comment is relevant to the context.

130s, 2022-07-15:
@cst0 Great! First would you helping out by joining the discussion that needs opinions/approvals https://github.com/AndreaCensi/csm/issues/33#issuecomment-1185129291?

130s, 2023-02-19:
@ubuntuslave If you're still around, I have 2 favors to ask. Would you mind:
- changing the default branch to `ros1`?
- adding @cst0 write access?
   - @cst0 Sorry took me awhile to respond. Are you still up for joining maintainership?

cst0, 2023-02-20:
> Are you still up for joining maintainership?

Sure!

130s, 2023-02-24:
Thanks @cst0!

> @ubuntuslave If you're still around, I have 2 favors to ask. Would you mind:
> - changing the default branch to `ros1`?
> - adding @cst0 write access?

@ubuntuslave Can you also give @marcalban write access? He's recently contributed PRs that require in-depth understanding of the codebase, and he agreed offline to co-maintain the repo.

I have a write access but don't have a maintainer access on this repo, so if we don't hear back and don't have these requests addressed, we might have to fork the repo somewhere else so that the new maintainers would have maintainer access (which happens a lot in OSS world, and original authors' credit remains in a new repo).

130s, 2023-02-24:
Thanks @cst0!

> @ubuntuslave If you're still around, I have 2 favors to ask. Would you mind:
> - changing the default branch to `ros1`?
> - adding @cst0 write access?

@ubuntuslave Can you also give @malban write access? He's recently contributed PRs that require in-depth understanding of the codebase, and he agreed offline to co-maintain the repo.

(I have a write access but don't have a maintainer access on this repo, so if we don't hear back and don't have these requests addressed, we might have to fork the repo somewhere else so that the new maintainers would have maintainer access (which happens a lot in OSS world, and original authors' credit remains in a new repo)).

cst0, 2023-05-23:
Great!

130s, 2023-06-02:
No response. I emailed the lab's PI, Dr. Xiao.

ehoxha91, 2023-06-12:
Hi, 

I'll give you guys write access. Sorry for not checking this earlier. However, it will take some time. I have permission issues myself, whenever I'm using another PC outside CCNY Robotics Lab. Thus, give me some time to go in the lab (I work from home), and I'll take care of this. 

-- Ejup

130s, 2023-06-12:
@ehoxha91 Thanks, I now do see "Settings" menu, however, inside of it, I see very minimum menus. E.g. I need to see "branch" menu (to change the default branch) but I don't see it. I also don't see the list of other maintainers.

Looking at [Repository roles for an organization](https://docs.github.com/en/organizations/managing-user-access-to-your-organizations-repositories/repository-roles-for-an-organization) page, "maintainer" falls short for the maintenance duties we're talking about. I guess we'll need "admin". Would you mind considering that?

130s, 2023-07-01:
Waiting for approval for full admin privilege.

Meanwhile @cst0  @malban you two also have limited maintainer access. Maybe update maintainer section in package.xml?

130s, 2023-10-04:
I asked one last time. If I don't hear back I'll suggest plan-B (fork and move the upstream there).

ehoxha91, 2023-10-06:
Hello everyone, will handle this today.

ehoxha91, 2023-10-06:
Isaac, your role is updated to Admin.

130s, 2023-10-10:
@ehoxha91 Thank you so much, I confired that I have maintenance access.

@cst0 @malban I gave you two a maintenance access as well. Would either of you mind updating `package.xml` (which ros.org refers to as the source of maintainers' contact info) with your info? With that done I think we can close this ticket IMO.

I don't foreesee actively using nor maintaining this repo for now. I can be a backup (like I did to take care of maintenance status) but I'd like to count on either of you for the maintenance if you don't mind.