CCNYRoboticsLab/scan_tools #75: ROS2 port

malban, 2022-07-12:
@130s 

I'm working on a ROS2 port of the laser_scan_matcher package that I would like to get pulled into this repo, if possible, and setup for releasing:
 - https://github.com/malban/scan_tools/tree/ros2

In addition to porting to ROS2, this version also:
 - fixes covariance calculation to capture the correct values and rotate them correctly into the odom frame
 - adds optional covariance augmentation based on detecting if there is a degenerate axis in the match, such as down a featureless hallway
 - removes velocity inputs and replaces them with a tf subscription for pose prediction.  It's assumed that a proper state estimator like [robot_localization](http://docs.ros.org/en/api/robot_localization/html/index.html) can be used to convert velocities into a predicted pose.
 - enables dynamic reconfiguration for most parameters

Related:
- https://github.com/AndreaCensi/csm/pull/34

130s, 2022-07-12:
Sounds good to me.

Changes you're proposing sounds like technically there are 2 different kinds: ROS2 porting, and fix/improvement. The latter can go to ROS1 branches. It'd ideally if you could split PRs, but that's up to you. At minimum, would you please make sure commits are separate for those 2 types of changes (so that others may easily cherry-pick if they want to)?

malban, 2022-07-14:
@130s I created PRs to the ros1 branch to:
- address the covariance issue: https://github.com/CCNYRoboticsLab/scan_tools/pull/78
- address the incorrect predictions and add tf as a prediction input: https://github.com/CCNYRoboticsLab/scan_tools/pull/79

prasun2712, 2023-10-04:
@malban I am not able to build the scan matcher package for ROS2 (https://github.com/malban/scan_tools/tree/ros2), looks like csm related error. It is not able to locate csm package, even though I have built the ROS2 csm package. Can you suggest on what might be the issue?