CCNYRoboticsLab/scan_tools #76: License declaration is missing

CsatariGergely, 2022-07-12:
The license declaration of the project is missing either in a form of a single LICENSE file or in a [Reuse](https://reuse.software/) compatible way. Declaration of the license makes the code more easy to consume and contribute to. Currently only the [ROS page](http://wiki.ros.org/laser_scan_matcher) of the project mentions BSD and LGPLv3, without specifying the variant of BSD and some of the source files contain BSD-3-Clause license headers.

130s, 2022-07-14:
Thanks for the help! The Open Robotics team also requested clarifying the license status https://github.com/ros/rosdistro/pull/28331#pullrequestreview-635987371, so this might as well have to be officially addressed.

130s, 2022-07-15:
@CsatariGergely While I agree to clarify license in a computer-friendly way (believe me as I've even started automating some part of free s/w licensing https://github.com/ros-infrastructure/rospkg/pull/201), to give a quick response in an interest of time,
- Overall all the software on this repo are likely BSD (Disclaimer: I'm relying on the declaration in `package.xml` files. I haven't yet done systematic analysis).
- `LGPLv3` might be coming from the usage of `GSL` in [`CSM`](https://github.com/AndreaCensi/csm), the library `laser_scan_matcher` package depends on.
   - There's a discussion going on right now to get rid of GSL https://github.com/AndreaCensi/csm/issues/33#issuecomment-1185129291. If that realizes, then we should be free from `LGPLv3`. But I assume merging all the suggested changes might take a week or even longer than that depending on how quick the conversation moves.
      - If you can't wait, you can manually build `laser_scan_matcher` against a WIP branch of CSM, either [csm_eigen](https://github.com/AndreaCensi/csm/tree/csm_eigen) (very outdated) or an open PR https://github.com/AndreaCensi/csm/pull/34 that spins off of the mentioned branch.

cst0, 2022-07-15:
I'm not sure LGPL is coming from `GSL`: the integration of GSL (a GPLv3 package) means that those branches are actually GPLv3 through copylefting, the license in the package/github page is (I believe) incorrectly reflecting the more permissive LGPL. I think I remember reading that the original implementation was released under LGPL intentionally, but I can't find any proof of that.

malban, 2022-07-16:
@cst0   I agree.  It seems like this repo was originally released and is currently licensed as LGPLv3, intentionally, regardless of the GSL dependency with is GPL.   

@130s  Make sure to note the difference between LGPL and GPL.

I think in practice, the GSL dependency makes downstream users of this library also need to conform to the GPL requirements from GSL (copyleft), and not just the lesser requirements of LGPL that this library is licensed under.   This is not inconsistent, but it is, I think, somewhat confusing and not clear to downstream users.

I don't think the license for this repo can be changed from LGPLv3 without explicit action from the original copyright holders.  It's also not necessary, because LGPLv3 is much less restrictive than GPL and is generally acceptable for most commercial use cases as the copyleft requirements are limited to modifying this library and don't effect things that only link to it.

The csm_eigen branch, which is also LGPLv3 removes the GSL (GPL) dependency, so downstream users won't need to link to GSL and therefore will only need to conform to the lesser LGPLv3 requirements, which should be in line with the normal ROS ecosystem.

CsatariGergely, 2022-07-18:
Thanks for the answers. I'm happy to see that others also would like to clarify the licensing in the repo. 
On "BSD" you mean BSD-3-Clause, BSD-2-Clause or some other variants of BSD?

malban, 2022-07-18:
Most commonly in ROS it's BSD-3-Clause.

It seems like that's the case here: https://github.com/CCNYRoboticsLab/scan_tools/blob/indigo/laser_scan_matcher/include/laser_scan_matcher/laser_scan_matcher.h#L1-L28

From what I can tell, all of the packages in this repo, except `polar_scan_matcher` are marked as BSD int he `package.xml` and contain source files with a BSD-3-Clause license.

The `laser_scan_matcher` `package.xml` also mentions `LGPLv3`, which I think is referring to the `csm` library dependency, but regardless, all of the source files are documented with a BSD-3-Clause license.

The `polar_scan_matcher` package is marked as GPLv2 and that is also documented in the source files within that package: https://github.com/CCNYRoboticsLab/scan_tools/blob/indigo/polar_scan_matcher/COPYING

malban, 2022-07-18:
I made a MR to address this: https://github.com/CCNYRoboticsLab/scan_tools/pull/80

130s, 2022-08-05:
Thanks for discussion.

- If my concern https://github.com/CCNYRoboticsLab/scan_tools/pull/80#pullrequestreview-1063923986 is correct, I think of 2 actions that can happen in parallel.
   - let's embrace GPL.
   - ask original authors to ask if it's ok to change license to what we think appropriate (I've seen it happening elsewhere).
- As discussed in https://github.com/CCNYRoboticsLab/scan_tools/pull/81, if we shift `laser_scan_matcher` to the upcoming eigen version of CSM we'll be able to remove (L/)GPL dependency, so those who want non (L/)GPL may just wait for that?

malban, 2022-08-05:
@130s I think the only package that is actually GPL is `polar_scan_matcher`

The rest are all BSD-3-Clause, with source files marked as such.

Additionally, laser_scan_matcher also calls out LGPL in the package.xml.  It's not clear what the intent of the LGPL reference is It could be:
  1. A dual license for the laser_scan_matcher package
  2. Referring to the fact that a core dependency, csm, is LGPL
  3. A mistake, since it was added without comment when the package was converted to catkin

For all practical purposes it's not important, with a dual license the end user is free to choose BSD if they want.   Additionally LGPL is not generally a problem for commercial use since it allows linking to LGPL libraries without triggering the copyleft terms.

So, I don't think there is any need to try and change the license(s) of this repo.

> if we shift laser_scan_matcher to the upcoming eigen version of CSM we'll be able to remove (L/)GPL dependency, so those who want non (L/)GPL may just wait for that?

Having said the above, while I don't think the licensing for this repo doesn't need to change,  it's not obvious from the current documentation that a core dependency to `laser_scan_matcher` is [csm](https://github.com/AndreaCensi/csm), which does have licensing implications for downstream users.

csm is LGPL, **which is generally not a problem for commercial use cases** since LGPL libraries can be linked against without triggering the copyleft clauses.

**Importantly**, there are two branches of csm, one that uses GSL and one that uses eigen.  **GSL is GPLv3**, which applies even when linking, **so this does** have implications to downstream users of laser_scan_matcher if it's linked against that version of csm.  Though, again, those copyleft requirements likely end at the node process boundary in ROS.

The eigen branch doesn't have the same issues since eigen is more permissive: https://eigen.tuxfamily.org/index.php?title=Main_Page#License

MR #80 tries to clarify this.

Ideally, we can get the ROS binary package of CSM to use the eigen branch, which will avoid confusion with the GPL dependency altogether.

130s, 2023-02-19:
https://github.com/CCNYRoboticsLab/scan_tools/pull/80 is merged. Please see if that resolves the issue.