CCNYRoboticsLab/scan_tools #84: I want to install it from source for Noetic

Baguviks, 2022-08-16:
I can install it from binary but I have to see the code and change somethings. When I install from source I receive an error:
```

/home/enis/catkin_ws/src/scan_tools-ros1/laser_scan_matcher/src/laser_scan_matcher.cpp: In member function ‘void scan_tools::LaserScanMatcher::processScan(laser_data*&, const ros::Time&)’:
/home/enis/catkin_ws/src/scan_tools-ros1/laser_scan_matcher/src/laser_scan_matcher.cpp:525:40: error: ‘f2b_kf_’ was not declared in this scope
  525 |       auto rotation = getLaserRotation(f2b_kf_);
      |                                        ^~~~~~~
Scanning dependencies of target laser_scan_splitter_nodelet
/home/enis/catkin_ws/src/scan_tools-ros1/laser_scan_matcher/src/laser_scan_matcher.cpp: In member function ‘Eigen::Matrix2f scan_tools::LaserScanMatcher::getLaserRotation(const tf::Transform&) const’:
/home/enis/catkin_ws/src/scan_tools-ros1/laser_scan_matcher/src/laser_scan_matcher.cpp:862:46: error: ‘laser_to_base_’ was not declared in this scope; did you mean ‘laser_from_base_’?
  862 |   tf::Transform laser_in_fixed = odom_pose * laser_to_base_;

Invoking "make -j8 -l8" failed
```

You have any idea how can i fix my problem?

cst0, 2022-08-17:
Hi, in the last merge there was some variables that got renamed, which is causing the error you're seeing here. It's discussed in #82, you can see the two lines that you'd have to change there.

130s, 2023-02-19:
As mentioned in https://github.com/CCNYRoboticsLab/scan_tools/issues/84#issuecomment-1218464318 some fixes were added and merged recently, so please try with the latest.

Closing for now.

closed 2023-02-19 by 130s