CCNYRoboticsLab/scan_tools #92: [Opinion] [laser_scan_matcher] Variable name is misleading

5730289021-NN, 2023-08-23:
I'm trying to understand the code. But I see that the code might confused someone in the future.

https://github.com/CCNYRoboticsLab/scan_tools/blob/7d5fe75b062f1b412d37fb2143ea33e776d83a73/laser_scan_matcher/src/laser_scan_matcher.cpp#L460-L470

On line 470, I think it should be
`tf::Transform pred_keyframe_laser_offset = base_from_laser_ * pred_keyframe_base_offset * laser_from_base_; `
instead.

`base_from_laser_` should mean to the "transformation from laser frame to base frame"

But on line 774,
https://github.com/CCNYRoboticsLab/scan_tools/blob/7d5fe75b062f1b412d37fb2143ea33e776d83a73/laser_scan_matcher/src/laser_scan_matcher.cpp#L774

I think the output of lookupTransform method doesn't provide the right meaning of `base_from_laser_`.

5730289021-NN, 2023-08-27:
There were people talking about this.
https://answers.ros.org/question/229463/confusing-tf-transforms/

Even the tf_echo uses in switched order
`echoListener.tf.waitForTransform(source_frameid, target_frameid, ros::Time(), ros::Duration(1.0));
`
where source is arg1 and target is arg2
https://github.com/ros/geometry/blob/521d9351d3742e7a24893ceacfd5fe92ad509763/tf/src/tf_echo.cpp#L102C56-L102C56

cst0, 2023-08-28:
Well, I certainly agree that the `tf` core concepts and implementation leave a lot of room for confusion. But in ROS answers post you've linked, what I'm really seeing is that people are acknowledging the confusion and actually calling out tf_echo as the outlier! And from the tf_echo source, the goal is to get the transform `from target_frame into the source_frame`: so even if they flip source/target, it looks like it's `arg1_from_arg2`.

5730289021-NN, 2023-08-31:
Ahh, I see what you meant.

When I see the variable name x_from_y, my head are just thinking out the `x referenced from y` which is opposite with the definition of yours.
(x: the coordinate frame of consideration, y: the coordinate frame of the origin)

That's fine. Thanks for your time.